enclustra-bsp/bsp-xilinx #2: Rootfile system

schegde, 2017-03-20:
How could i change the rootfile system from busybox to ubuntu 14.04 corefs for ARM?

niklas88, 2017-04-26:
This is a bit hacky but works for me. Simply replace the contents of the root file system with those from a similar system. I've successfully done this with the Debian from a BeagleBone Black. Note that you might need to be careful around the whole hard- vs softfload difference.

schegde, 2017-05-01:
I tried that, but throws up a "no Working init found" error. So I did a manual kernel header and module build and installed them on Ubuntu 14.04 ARM file system, and that worked with the boot files generated from enclustra-BSP.Need to make sure the kernel being used with this file system is of same version or higher than the headers. Thanks for the help!

closed 2017-05-01 by schegde