enclustra-bsp/bsp-xilinx #6: host-e2fsprogs 1.43.3 Building conflict with glibc

j4munoz, 2018-05-30:
Probably using the latest e2fsprogs will solve it. Here is the compilation error that I got:

>>> host-e2fsprogs 1.43.3 Building 
PATH="/home/dev/bsp-xilinx/sources/buildroot-rootfs/output/host/bin:/home/dev/bsp-xilinx/sources/buildroot-rootfs/output/host/sbin:/home/dev/bsp-xilinx/sources/buildroot-rootfs/output/host/usr/bin:/home/dev/bsp-
xilinx/sources/buildroot-rootfs/output/host/usr/sbin:/home/dev/bsp-xilinx/bin/arm-none-linux-gnueabi-static/bin:/home/dev/bsp-xilinx/bin/aarch64-none-linux-gnueabi/bin:/home/dev/bsp-xilinx/bin/uboot-tools-i686-s
tatic:/home/dev/bsp-xilinx/bin/device-tree-compiler-i686-static:/home/dev/bsp-xilinx/bin/mkbootimage:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin" PKG_CONFIG
="/home/dev/bsp-xilinx/sources/buildroot-rootfs/output/host/usr/bin/pkg-config" PKG_CONFIG_SYSROOT_DIR="/" PKG_CONFIG_ALLOW_SYSTEM_CFLAGS=1 PKG_CONFIG_ALLOW_SYSTEM_LIBS=1 PKG_CONFIG_LIBDIR="/home/dev/bsp-xilinx/
sources/buildroot-rootfs/output/host/usr/lib/pkgconfig:/home/dev/bsp-xilinx/sources/buildroot-rootfs/output/host/usr/share/pkgconfig" /usr/bin/make -C /home/dev/bsp-xilinx/sources/buildroot-rootfs/output/buil
d/host-e2fsprogs-1.43.3/ 
...
CC create_inode.o
./../misc/create_inode.c: In function ‘set_inode_xattr’:
./../misc/create_inode.c:136:9: warning: implicit declaration of function ‘llistxattr’; did you mean ‘lstat64’? [-Wimplicit-function-declaration]
size = llistxattr(filename, NULL, 0);
^~~~~~~~~~
lstat64
./../misc/create_inode.c:172:16: warning: implicit declaration of function ‘lgetxattr’; did you mean ‘getdate_r’? [-Wimplicit-function-declaration]
value_size = lgetxattr(filename, name, NULL, 0);
^~~~~~~~~
getdate_r
./../misc/create_inode.c: In function ‘do_mknod_internal’:
./../misc/create_inode.c:290:13: warning: In the GNU C Library, "major" is defined
by <sys/sysmacros.h>. For historical compatibility, it is
currently defined by <sys/types.h> as well, but we plan to
remove this soon. To use "major", include <sys/sysmacros.h>
directly. If you did not intend to use a system-defined macro
"major", you should undefine it after including <sys/types.h>.
devmajor = major(st->st_rdev);
^~~~~~~~~~~~~~~~~~~~
./../misc/create_inode.c:291:13: warning: In the GNU C Library, "minor" is defined
by <sys/sysmacros.h>. For historical compatibility, it is
currently defined by <sys/types.h> as well, but we plan to
remove this soon. To use "minor", include <sys/sysmacros.h>
directly. If you did not intend to use a system-defined macro
"minor", you should undefine it after including <sys/types.h>.
devminor = minor(st->st_rdev);
^~~~~~~~~~~~~~~~~~~~
./../misc/create_inode.c: At top level:
./../misc/create_inode.c:395:18: error: conflicting types for ‘copy_file_range’
static errcode_t copy_file_range(ext2_filsys fs, int fd, ext2_file_t e2_file,
^~~~~~~~~~~~~~~
In file included from ./../misc/create_inode.c:19:0:
/usr/include/unistd.h:1110:9: note: previous declaration of ‘copy_file_range’ was here
ssize_t copy_file_range (int _infd, __off64_t *_pinoff,
^~~~~~~~~~~~~~~
Makefile:418: recipe for target 'create_inode.o' failed
make[4]: *** [create_inode.o] Error 1
Makefile:405: recipe for target 'all-progs-recursive' failed
make[3]: *** [all-progs-recursive] Error 1
Makefile:340: recipe for target 'all' failed
make[2]: *** [all] Error 2
package/pkg-generic.mk:216: recipe for target '/home/dev/bsp-xilinx/sources/buildroot-rootfs/output/build/host-e2fsprogs-1.43.3/.stamp_built' failed
make[1]: *** [/home/dev/bsp-xilinx/sources/buildroot-rootfs/output/build/host-e2fsprogs-1.43.3/.stamp_built] Error 2
Makefile:76: recipe for target '_all' failed
make: *** [_all] Error 2
ERROR: Error running make EBE_OVERLAYS=
{ebe_overlays} 
-j9 for Buildroot

alehed, 2018-07-04:
This is caused by a newer glibc (2.27) defining `copy_file_range`. This issue was fixed in e2fsprogs v1.43.8. There are two possible fixes:

1. Update buildroot (to at least version 2018.2 which has e2fsprogs v1.43.9)
2. Manually apply the patch [01551bdba16ab16512a01affe02ade32c41ede8a](https://github.com/tytso/e2fsprogs/commit/01551bdba16ab16512a01affe02ade32c41ede8a) by adding another .patch to the package/e2fsprogs directory.

The first option is probably better.

alehed, 2018-07-22:
For reference, the patch I put, into `sources/buildroot-rootfs/package/e2fsprogs/` as a temporary workaround is:

```
This patch prevents a conflict with glibc

--- a/misc/create_inode.c
+++ b/misc/create_inode.c	
@@ -392,7 +392,7 @@ static ssize_t my_pread(int fd, void *buf, size_t count, off_t offset)
 }
 #endif /* !defined HAVE_PREAD64 && !defined HAVE_PREAD */
 
-static errcode_t copy_file_range(ext2_filsys fs, int fd, ext2_file_t e2_file,
+static errcode_t copy_file_chunk(ext2_filsys fs, int fd, ext2_file_t e2_file,
 				 off_t start, off_t end, char *buf,
 				 char *zerobuf)
 {
@@ -466,7 +466,7 @@ static errcode_t try_lseek_copy(ext2_filsys fs, int fd, struct stat *statbuf,
 
 		data_blk = data & ~(fs->blocksize - 1);
 		hole_blk = (hole + (fs->blocksize - 1)) & ~(fs->blocksize - 1);
-		err = copy_file_range(fs, fd, e2_file, data_blk, hole_blk, buf,
+		err = copy_file_chunk(fs, fd, e2_file, data_blk, hole_blk, buf,
 				      zerobuf);
 		if (err)
 			return err;
@@ -516,7 +516,7 @@ static errcode_t try_fiemap_copy(ext2_filsys fs, int fd, ext2_file_t e2_file,
 		}
 		for (i = 0, ext = ext_buf; i < fiemap_buf->fm_mapped_extents;
 		     i++, ext++) {
-			err = copy_file_range(fs, fd, e2_file, ext->fe_logical,
+			err = copy_file_chunk(fs, fd, e2_file, ext->fe_logical,
 					      ext->fe_logical + ext->fe_length,
 					      buf, zerobuf);
 			if (err)
@@ -569,7 +569,7 @@ static errcode_t copy_file(ext2_filsys fs, int fd, struct stat *statbuf,
 		goto out;
 #endif
 
-	err = copy_file_range(fs, fd, e2_file, 0, statbuf->st_size, buf,
+	err = copy_file_chunk(fs, fd, e2_file, 0, statbuf->st_size, buf,
 			      zerobuf);
 out:
 	ext2fs_free_mem(&zerobuf);

```

j4munoz, 2018-07-31:
Thank you, the patch is working. Alternatively, Using Debian 9.5 (stretch) also works out of the box.

closed 2018-07-31 by j4munoz

jiejieTop, 2019-12-04:
> For reference, the patch I put, into `sources/buildroot-rootfs/package/e2fsprogs/` as a temporary workaround is:
>
> ```
> This patch prevents a conflict with glibc
>
> --- a/misc/create_inode.c
> +++ [...]

Thank you，the patch is working，爱你哦~

apaj, 2020-03-20:
Works for me, as well. Thanks a lot.

Hassan313, 2020-07-06:
@alehed Hi, Thank you very much. You saved my day. Again, thank you very much for writing the solution for others to use it.

thinhx2, 2020-07-08:
> This is caused by a newer glibc (2.27) defining `copy_file_range`. This issue was fixed in e2fsprogs v1.43.8. There are two possible fixes:
>
> 1. Update buildroot (to at least version 2018.2 which has [...]

thank you, it working