enclustra-bsp/bsp-xilinx #10: XCZU15EG shows only 2GiB of RAM instead of 4GiB

j4munoz, 2019-03-15:
XCZU15EG shows only 2GiB of RAM instead of 4GiB. I am not sure if the problem comes "just" from the Device Tree, or also from the bitstream of configuration for the UltraScale+.

tgorochowik, 2019-03-20:
Hello @j4munoz 

The problem is that there are two variants of the devices, some of them with 2GiB and some of them with 4GiB of ram, and we currently ship common code and binaries for both of them.

If you want to change that manually, you could change the devicetree like so:
For U-Boot:
```
diff --git a/arch/arm/dts/mercury-xu1.dts b/arch/arm/dts/mercury-xu1.dts
index 5e81e06..ef86a7e 100644
--- a/arch/arm/dts/mercury-xu1.dts
+++ b/arch/arm/dts/mercury-xu1.dts
@@ -42,7 +42,7 @@
 
        memory {
                device_type = "memory";
-               reg = <0x0 0x0 0x0 0x80000000>;
+               reg = <0x0 0x0 0x0 0x80000000>, <0x8 0x00000000 0x0 0x80000000>; /* 4GB */
        };
```

And for Linux:
```
diff --git a/arch/arm64/boot/dts/xilinx/zynqmp-enclustra-xu1.dts b/arch/arm64/boot/dts/xilinx/zynqmp-enclustra-xu1.dts
index 43b3a9c7..65095f0 100644
--- a/arch/arm64/boot/dts/xilinx/zynqmp-enclustra-xu1.dts
+++ b/arch/arm64/boot/dts/xilinx/zynqmp-enclustra-xu1.dts
@@ -45,7 +45,7 @@
 
        memory {
                device_type = "memory";
-               reg = <0x0 0x0 0x0 0x80000000>;
+               reg = <0x0 0x0 0x0 0x80000000>, <0x8 0x00000000 0x0 0x80000000>; /* 4GB */
        };
```

To make it use all the memory.

Note that we are now evaluating the possibilities to detect the amount of RAM properly (or to allow users to choose it), but we cannot give you a date for that yet.

j4munoz, 2019-03-21:
Thank you, it worked 👍 
Shall I close the issue?

tgorochowik, 2019-03-21:
Sure! I guess this can be closed, we will keep it in our internal bug tracker to see if the automatic detection can be implemented.

Thanks!

j4munoz, 2019-03-21:
Closing then, thank you for your support @tgorochowik !

closed 2019-03-21 by j4munoz