enclustra-bsp/bsp-xilinx #11: Problems with SD boot on Mercury XU5 and Mercury PE1-300

DNR1406, 2019-03-18:
Just started using the BSP to try and boot from the SD card slot on the PE1-300. Used the build script to generate the necessary output files and copied them to the SD card (Not UHS) and selected the right DIP switch configuration according to the documentation. 
However, it still doesn't seem to be working. I also tried to hook up the PE1-300 to a serial console to see if the boot process would be displayed there but no luck either. 
How do I know if the boot process has finished? 
(The files on the SD card are: boot.bin, Image, uboot.scr and devicetree.dtb).

skoster-enclustra, 2019-03-19:
Assuming your dip switches and voltage jumpers are correct and you're hooked up to the right usb port you should see the boot log over uart. Are you following the [guide ](http://enclustra.github.io/ebe-docs/user-doc-xilinx/index_xilinx.html#sd-card-mmc) to format the sd card? 

If you get stuck feel free to email enclustra support.

DNR1406, 2019-03-19:
Thanks for responding so quickly. Yes, I've followed the guide to format the sd card so that I have two partitions, one boot partition (where all the output files are placed) and one root partition. I hooked the serial console to the micro USB port on the PE1-300 assuming that this one is correct because the current DIP switch configuration routes the FTDI signals to the micro USB port. The DIP switch settings are CFG A 0001 and CFG B 1011 (the leftmost bit being the first one). 
Am I missing something?

DNR1406, 2019-03-19:
As of right now, I'm using the voltage jumpers set to 1-2 and 9-10 but this configuration sends 3.3V to the module connectors and not 1.8v. Would you mind telling me how to configure the voltage jumpers to be set at 1.8v?

skoster-enclustra, 2019-03-19:
Can you tell me which module you're using?

DNR1406, 2019-03-19:
Yes, I'm using the Mercury XU5-2EG (XCZU2EG-1SFVC784I)

skoster-enclustra, 2019-03-19:
Try vselA= B and VSEL B= B.

DNR1406, 2019-03-19:
Ah yes, I see it now. I can configure the voltage jumpers to be at 1.8v as well. Thank you very much!

closed 2019-03-19 by DNR1406

zhangyeyan, 2019-09-17:
Hi,i meet the same problem,i use XCZU5EV-SFVC784AAZ , PE1 ,i also want to BOOT from SD and use mico usb 2.0 to connect to minicom,i use the default  DIP switch configuration (the default configuration  boots from  SD)(CFG A 1110  and CFG B 1111 ),i also try vselA= B and VSEL B= B,but it doesn't work,how can i solve the problem?

zhangyeyan, 2019-09-17:
I try again vselA= B and VSEL B= B ,use ttyUSB1 ,it works !!!!!Thank you !!!!!