enclustra-bsp/bsp-xilinx #20: [ZX1][PE1] Can't get into U-Boot with UART

faku99, 2020-03-03:
Hey,

I'm trying to boot Linux from the SD card on the ZX1 module using the PE1 board.

To do so, I cloned this repository and ran the `build.sh` script.
Then, I selected **U-Boot**, **Buildroot** and **Linux** as targets. The tool successfully compiled U-Boot (Linux compilation failed but that's not what this issue is about).

Then, I prepared an SD card following the [Xilinx guide](https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/18841655/Prepare+Boot+Medium). I copied the freshly compiled `boot.bin` to the first partition (`BOOT`) of the SD card and then ejected it.

Once the SD card was plugged into the PE1 board, I configured the board as follow:
- **CFG A**: `0001`
- **CFG B**: `1000`
- **VSEL A**: `3V3`
- **VSEL B**: `3V3`

When I switch on the board, the `PWGD` LED becomes green, the `CPU` LED yellow and the `SC`LED blinks in yellow. However, when I run `picocom -b 115200` on both `ttyUSB0` and `ttyUSB1`, there is not output.

I think I followed exactly what the documentation and the guide say. Am I missing something?

faku99, 2020-03-05:
Hello? Is there another place where I can file this issue?

abuerkler, 2020-03-05:
Dear Lucas

The Linux build environment uses bitstreams and PS configurations of the Vivado projects from the reference designs that can be downloaded from the Enclustra's download page. The correct settings for the DIP switches and voltage jumpers can be seen in the manual of the ZX1 reference design.

VSEL A: 2.5V (position A)
VSEL B: 1.8V (position B)

CFG A = [1: OFF, 2: OFF, 3: OFF, 4: ON ]
CFG B = [1: OFF, 2: OFF, 3: ON, 4: ON]

VCC_IO_B supports only max. 1.8V! To prevent the ZX1 module from being damaged, VCC_IO_B is only enabled if it is below 2.2V. The module is also held in reset if the voltage is too big. That can be seen when the CPU LED is enabled.

I recommend you to write directly to support@enclustra.com to get an answer faster.

Best regards,
Andreas Buerkler

faku99, 2020-03-05:
Hey Andres,
Thank you very much for your answer. I managed to get output from the UART.

For your information, I wrote directly to the email address you mentioned but didn't get an answer either.

Again, thank you very much!

closed 2020-03-05 by faku99