enclustra-bsp/bsp-xilinx #21: ERROR: Error running make EBE_OVERLAYS={ebe_overlays} -j9 for Buildroot

wangyuanzhengbighead, 2020-04-10:
I use build.sh to build up Zynq-Ultrascale+_Mercury_XU7_Mercury_PE1_EMMC, uboot，rootf and kernel,but always failed, what's the problem?   here is my log
[log.txt](https://github.com/enclustra-bsp/bsp-xilinx/files/4462512/log.txt)

abuerkler, 2020-04-14:
Hello wangyuanzhengbighead

It looks like building Buildroot is failing. There is a detailed log file "bsp-xilinx/build.log" that shows more detailed information. Please send this file to support@enclustra.com.

Best regards

LeFl0w, 2020-11-09:
Hi,
Same error here (on Windows subsystem on linux : Ubuntu 18.04LTS)
The buil.log file showed error : 
```
sh: 1: autom4te: not found
aclocal.tmp: error: echo failed with exit status: 127
```
even though automake was installed.
This was fixed by adding the following packages : 

`sudo apt-get install libtool intltool`

Simon-MS-MIS, 2022-01-13:
Hello,
I'm getting the same error. Just before that, in the log file there are 3 unsuccessful attempts to connect to ftp://ftp.infradead.org/pub/mtd-utils/mtd-utils-2.1.1.tar.bz2 althought I manually installed mtd-utils via apt.

I'm using VM Ubuntu 18.04.6 LTS.

tholzsche, 2022-01-17:
Hi, I hope this is just a temporary problem with the server, but the following quick fix should solve the issue:
```
diff --git a/package/mtd/mtd.mk b/package/mtd/mtd.mk
index 3477460..34e0b39 100644
--- a/package/mtd/mtd.mk
+++ b/package/mtd/mtd.mk
@@ -6,7 +6,7 @@
 
 MTD_VERSION = 2.1.1
 MTD_SOURCE = mtd-utils-$(MTD_VERSION).tar.bz2
-MTD_SITE = ftp://ftp.infradead.org/pub/mtd-utils
+MTD_SITE = https://infraroot.at/pub/mtd
 MTD_LICENSE = GPL-2.0
 MTD_LICENSE_FILES = COPYING
 MTD_INSTALL_STAGING = YES
```

florianlederer, 2022-03-09:
I've got a simlar error: 

<pre><code>ERROR: Error running make EBE_OVERLAYS={ebe_overlays} -j49 for Buildroot</code></pre>

build.log:
<pre><code>>>> mmc-utils d40ec535b9d4e4c974e8c2fbfb422cd0348cc5e8 Downloading
Initialized empty Git repository in /root/delete/bsp-xilinx/sources/buildroot-rootfs/dl/mmc-utils/git/.git/
Fetching all references
fatal: unable to connect to git.kernel.org:
git.kernel.org[0: 145.40.68.75]: errno=Connection timed out
git.kernel.org[1: 2604:1380:4601:e00::1]: errno=Cannot assign requested address

Detected a corrupted git cache.
Removing it and starting afresh.
Initialized empty Git repository in /root/delete/bsp-xilinx/sources/buildroot-rootfs/dl/mmc-utils/git/.git/
Fetching all references
fatal: unable to connect to git.kernel.org:
git.kernel.org[0: 145.40.68.75]: errno=Connection timed out
git.kernel.org[1: 2604:1380:4601:e00::1]: errno=Cannot assign requested address

Detected a corrupted git cache.
This is the second time in a row; bailing out
package/pkg-generic.mk:167: recipe for target '/root/delete/bsp-xilinx/sources/buildroot-rootfs/output/build/mmc-utils-d40ec535b9d4e4c974e8c2fbfb422cd0348cc5e8/.stamp_downloaded' failed
make: *** [/root/delete/bsp-xilinx/sources/buildroot-rootfs/output/build/mmc-utils-d40ec535b9d4e4c974e8c2fbfb422cd0348cc5e8/.stamp_downloaded] Error 1
Error running make EBE_OVERLAYS={ebe_overlays} -j49 for Buildroot
Getting binary boot_files_ME-XU5-3EG-2I-D11E</code></pre>

It looks like _git://git.kernel.org/pub/scm/linux/kernel/git/cjb/mmc-utils.git_ is not available anymore.

The following solved it for me:
<pre><code>diff --git a/package/mmc-utils/mmc-utils.mk b/package/mmc-utils/mmc-utils.mk
index 8c264a4..53002bb 100644
--- a/package/mmc-utils/mmc-utils.mk
+++ b/package/mmc-utils/mmc-utils.mk
@@ -5,7 +5,7 @@
 ################################################################################
 
 MMC_UTILS_VERSION = d40ec535b9d4e4c974e8c2fbfb422cd0348cc5e8
-MMC_UTILS_SITE = git://git.kernel.org/pub/scm/linux/kernel/git/cjb/mmc-utils.git
+MMC_UTILS_SITE =  http://sources.buildroot.net/mmc-utils
 MMC_UTILS_LICENSE = GPL-2.0
 MMC_UTILS_LICENSE_FILES = mmc.h</code></pre>

abuerkler, 2022-03-14:
> It looks like _git://git.kernel.org/pub/scm/linux/kernel/git/cjb/mmc-utils.git_ is not available anymore.

Thank you for the notification. Cloning this repository seems to be working. I guess this was only a temporary issue and the server was not available at that time.

21576181, 2024-03-26:
Hi, I am having the same issue: Error running make EBE_OVERLAYS={ebe_overlays} -j17 for Buildroot

These are my parameters:
Device: Zynq-Ultrascale+ Mercury XU5 Mercury PE1 MMC 
Targets: U-Boot (fetch + build), Buildroot (fetch), Linux (fetch + build) 
Binaries: ME-XU5-5EV-2I-D12E 

Here is my log file: 
[build.log](https://github.com/enclustra-bsp/bsp-xilinx/files/14755491/build.log)

The above solutions did not work for me.