enclustra-bsp/bsp-xilinx #22: Which version of Python3 did you test with? Error in Python 3.6.9

mbanders, 2020-05-07:
The readme says "compatible both with Python 2 and Python 3", so I'm trying some things in python3:

```sh
buildscripts$ python --version
Python 3.6.9

buildscripts$ ./build.py -l
Traceback (most recent call last):
  File "./build.py", line 271, in <module>
    revision = utils.get_git_revision(bscripts_path).rstrip('\n')
TypeError: a bytes-like object is required, not 'str'
```

tholzsche, 2020-05-26:
Thank you for the information. It looks like it doesn't support Python 3 without a few modifications. I encountered the same problem (with several Python 3.x versions).
﻿

tholzsche, 2020-06-18:
This fix seems to work and will be included in the next release: https://github.com/enclustra-bsp/enclustra-buildscripts/pull/16

closed 2020-06-18 by tholzsche