enclustra-bsp/bsp-xilinx #24: usb 2.0 not work on Mercury+ PE1

wangyuanzhengbighead, 2020-05-29:
ths soc  board that i use is Mercury+ XU1 15eg the base board i use is Mercury+ PE1.  Linux generated from your script will boot normally,But there are no reflection of the usb2.0 port insert, such as Mouse and keyboard usb flash drive. and I don't see any print information by using the "dmesg" ,  What does it take to work properly?

[image]
[image]

abuerkler, 2020-05-29:
Dear wangyuanzhengbighead

If you are using the USB PHY0 on the XU1 (default in EBE), the DIP switches on the PE1 need to be set correctly to connect USB PHY0 to the USB hub on the PE1 base board (CFG B needs to be: ON OFF ON ON, see https://download.enclustra.com/public_files/Base_Boards/Mercury+_PE1/Mercury_PE1_User_Manual_V06.pdf, section 6.5 USB). The setting on your image is valid if you use the USB PHY1.

There is an issue on the PE1 base board in combination with the XU1 module. The data signals of the USB PHY0 are connected through multiplexers to the USB hub on the PE1, but to configure the multiplexer correctly, the USB_ID signal needs to be '0'. You can force the USB_ID signal to '0' by executing following I2C commands. This sets the corresponding pin of the system controller as output driving '0'.

i2cset -yf 0 0x0d 0x14 0x0
i2cset -yf 0 0x0d 0x15 0x1

Please note that the above commands must only be executed if USB PHY1 is in reset. If the USB PHY1 is in use you may damage the hardware. As alternative, you can use USB PHY1, which doesn't need this workaround.

Best regards,
Andreas Buerkler

wangyuanzhengbighead, 2020-06-17:
i want to Expand a display， but there are no hdmi or dp port, what should i do?
(that i use is Mercury+ XU1 15eg the base board i use is Mercury+ PE1.)

dduerner, 2020-06-18:
Hi wangyuanzhengbighead

As this question is not related to this ticket nor to the Linux itself, we would be happy to receive such requests via the Enclustra Support channel: [https://www.enclustra.com/en/support/]. If the answer above from Andreas Buerkler has resolved your issue regarding the USB, please close the ticket. 

Best regards, 
 Daniel Duerner

closed 2020-06-18 by wangyuanzhengbighead