enclustra-bsp/bsp-xilinx #26: UBOOT build fails with missing default configuration

pwittich, 2020-06-16:
I'm trying to build an image for a Mars XU3 with the EB1 base board. The UBOOT build fails as follows. This is in a docker image based on Ubuntu 14.04 LTS with the packages as described in the user documentation. 

The "Building Linux" step appears to be running ok.

Any help would be appreciated. 

From the script:
```
builduser@adc77121fb2e:/bsp-xilinx$ ./build.sh -d Zynq-Ultrascale+/Mars_XU3/Mars_EB1/MMC
INFO: Enclustra Build Environment (v1.8.2-f8ef009)
Running under Python version 2.7.6.

Copyright (c) 2015-2017 Enclustra GmbH, Switzerland.
All rights reserved.

INFO: Fetching U-Boot
+git submodule init xilinx-uboot
+git submodule update --remote --depth 1 xilinx-uboot
+git fetch --depth 1 origin v1.8.2
+git checkout FETCH_HEAD
Target U-Boot fetched
INFO: Fetching Linux
+git submodule init xilinx-linux
+git submodule update --remote --depth 1 xilinx-linux
+git fetch --depth 1 origin v1.8.2
+git checkout FETCH_HEAD
Target Linux fetched
INFO: Fetching Buildroot
+git submodule init buildroot-rootfs
+git submodule update --remote --depth 1 buildroot-rootfs
+git fetch --depth 1 origin v1.8.2
+git checkout FETCH_HEAD
Target Buildroot fetched
+curl -O http://www.enclustra.com/binaries/enclustra-bsp/v1.8.2/arm-none-linux-gnueabi-static-i686.tar.bz2
+curl -O http://www.enclustra.com/binaries/enclustra-bsp/v1.8.2/aarch64-none-linux-gnueabi-static-i686.tar.gz
+curl -O http://www.enclustra.com/binaries/enclustra-bsp/v1.8.2/uboot-tools-i686-static.tar.gz
+curl -O http://www.enclustra.com/binaries/enclustra-bsp/v1.8.2/device-tree-compiler-i686-static.tar.gz
+curl -O http://www.enclustra.com/binaries/enclustra-bsp/v1.8.2/mkbootimage-e3796c3.tar.gz
INFO: Building U-Boot
+make CROSS_COMPILE="aarch64-none-linux-gnueabi-" CROSS32CC="arm-none-linux-gnueabi-gcc" enclustra_ebe_generated_defconfig -j4
ERROR: Error running make CROSS_COMPILE="aarch64-none-linux-gnueabi-" CROSS32CC="arm-none-linux-gnueabi-gcc" enclustra_ebe_generated_defconfig -j4 for U-Boot
+make CROSS_COMPILE="aarch64-none-linux-gnueabi-" CROSS32CC="arm-none-linux-gnueabi-gcc" -j4
ERROR: Error running make CROSS_COMPILE="aarch64-none-linux-gnueabi-" CROSS32CC="arm-none-linux-gnueabi-gcc" -j4 for U-Boot
+cat configs/enclustra_zynqmp_mars_xu3_defconfig configs/enclustra_env_on_sd.appendix > configs/enclustra_ebe_generated_defconfig
ERROR: Error running cat configs/enclustra_zynqmp_mars_xu3_defconfig configs/enclustra_env_on_sd.appendix > configs/enclustra_ebe_generated_defconfig for U-Boot
+mkimage -A arm -O linux -T script -C none -a 0 -e 0 -n "Uboot mmc start script" -d bscripts/mmcboot-rootfs-us bscripts/uboot.scr
ERROR: Error running mkimage -A arm -O linux -T script -C none -a 0 -e 0 -n "Uboot mmc start script" -d bscripts/mmcboot-rootfs-us bscripts/uboot.scr for U-Boot
+mkimage -A arm -O linux -T script -C none -a 0 -e 0 -n "Uboot mmc ramdisk start script" -d bscripts/mmcboot-ramdisk-us bscripts/uboot_ramdisk.scr
ERROR: Error running mkimage -A arm -O linux -T script -C none -a 0 -e 0 -n "Uboot mmc ramdisk start script" -d bscripts/mmcboot-ramdisk-us bscripts/uboot_ramdisk.scr for U-Boot
```

From the build log:
```
make CROSS_COMPILE="aarch64-none-linux-gnueabi-" CROSS32CC="arm-none-linux-gnueabi-gcc" enclustra_ebe_generated_defconfig -j4
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  SHIPPED scripts/kconfig/zconf.tab.c
  SHIPPED scripts/kconfig/zconf.lex.c
  SHIPPED scripts/kconfig/zconf.hash.c
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
***
*** Can't find default configuration "arch/../configs/enclustra_ebe_generated_defconfig"!
***
make[1]: *** [enclustra_ebe_generated_defconfig] Error 1
make: *** [enclustra_ebe_generated_defconfig] Error 2
Error running make CROSS_COMPILE="aarch64-none-linux-gnueabi-" CROSS32CC="arm-none-linux-gnueabi-gcc" enclustra_ebe_generated_defconfig -j4 for U-Boot
make CROSS_COMPILE="aarch64-none-linux-gnueabi-" CROSS32CC="arm-none-linux-gnueabi-gcc" -j4
scripts/kconfig/conf  --silentoldconfig Kconfig
***
*** Configuration file ".config" not found!
***
*** Please run some configurator (e.g. "make oldconfig" or
*** "make menuconfig" or "make xconfig").
***
make[2]: *** [silentoldconfig] Error 1
make[1]: *** [silentoldconfig] Error 2
make: *** No rule to make target `include/config/auto.conf', needed by `include/config/uboot.release'.  Stop.
Error running make CROSS_COMPILE="aarch64-none-linux-gnueabi-" CROSS32CC="arm-none-linux-gnueabi-gcc" -j4 for U-Boot
```

tholzsche, 2020-06-18:
Hi,

Somehow it looks like the cause of the issue is based on the failure of the cat command. Can verify that the configuration files exist? And can you further supply the full build.log?

But I would suggest that you simply set up the docker image again and install the required packages as described in the docs (https://enclustra.github.io/ebe-docs/user-doc-xilinx/index_xilinx.html#ubuntu-14-04-lts) e.g.:
```
docker run -it ubuntu:trusty /bin/bash
~/# cat /etc/lsb-release 
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=14.04
DISTRIB_CODENAME=trusty
DISTRIB_DESCRIPTION="Ubuntu 14.04.6 LTS"
~/# apt update
~/# apt install -y autoconf bc build-essential curl git mercurial \
                    libc6-i386 python python-dialog unzip wget
~/# git clone https://github.com/enclustra-bsp/bsp-xilinx
~/#  cd bsp-xilinx
~/#  ./build.sh -d Zynq-Ultrascale+/Mars_XU3/Mars_EB1/MMC
...
INFO: Building U-Boot
+cat configs/enclustra_zynqmp_mars_xu3_defconfig configs/enclustra_env_on_sd.appendix > configs/enclustra_ebe_generated_defconfig
cat configs/enclustra_zynqmp_mars_xu3_defconfig configs/enclustra_env_on_sd.appendix > configs/enclustra_ebe_generated_defconfig completed successfully
+make CROSS_COMPILE="aarch64-none-linux-gnueabi-" CROSS32CC="arm-none-linux-gnueabi-gcc" enclustra_ebe_generated_defconfig -j7
make CROSS_COMPILE="aarch64-none-linux-gnueabi-" CROSS32CC="arm-none-linux-gnueabi-gcc" enclustra_ebe_generated_defconfig completed successfully
+make CROSS_COMPILE="aarch64-none-linux-gnueabi-" CROSS32CC="arm-none-linux-gnueabi-gcc" -j7
make CROSS_COMPILE="aarch64-none-linux-gnueabi-" CROSS32CC="arm-none-linux-gnueabi-gcc" completed successfully
+mkimage -A arm -O linux -T script -C none -a 0 -e 0 -n "Uboot mmc start script" -d bscripts/mmcboot-rootfs-us bscripts/uboot.scr
mkimage -A arm -O linux -T script -C none -a 0 -e 0 -n "Uboot mmc start script" -d bscripts/mmcboot-rootfs-us bscripts/uboot.scr completed successfully
+mkimage -A arm -O linux -T script -C none -a 0 -e 0 -n "Uboot mmc ramdisk start script" -d bscripts/mmcboot-ramdisk-us bscripts/uboot_ramdisk.scr
mkimage -A arm -O linux -T script -C none -a 0 -e 0 -n "Uboot mmc ramdisk start script" -d bscripts/mmcboot-ramdisk-us bscripts/uboot_ramdisk.scr completed successfully
INFO: Building Linux
...

```

Greets Till

pwittich, 2020-06-18:
Hi Till, Thanks for your response. I am attaching the full build log.
[build.log](https://github.com/enclustra-bsp/bsp-xilinx/files/4799353/build.log)
The docker was built with all those packages, but is a slightly older subversion of LTS 14.04:
```
builduser@adc77121fb2e:/bsp-xilinx$ cat /etc/lsb-release 
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=14.04
DISTRIB_CODENAME=trusty
DISTRIB_DESCRIPTION="Ubuntu 14.04.5 LTS"
```
Re that `cat` command failing, I can't really tell from where the command is being run, but a search of the file system suggests it doesn't exist.

pwittich, 2020-06-18:
I just deleted and re-did the `git clone` command and now it fails differently, see new build log also attached...

tholzsche, 2020-06-19:
I'm sorry this looks like a docker problem. Either a fault in docker itself or an incorrect usage. Maybe this could help: https://docs.docker.com/storage/volumes/