enclustra-bsp/bsp-xilinx #28: Buildroot fails to build in Debian 10

michael-vacek, 2020-06-18:
Hi,

after I migrated to Debian 10, the Enclustra Buildroot fails to build with the following errors:

`make[4]: Entering directory '..../enclustra/bsp-xilinx/sources/buildroot-rootfs/output/build/host-m4-1.4.18/lib'
  CC       freadahead.o
  CC       fseeko.o
freadahead.c: In function 'freadahead':
freadahead.c:92:3: error: #error "Please port gnulib freadahead.c to your platform! Look at the definition of fflush, fread, ungetc on your system, then report this to bug-gnulib."
  #error "Please port gnulib freadahead.c to your platform! Look at the definition of fflush, fread, ungetc on your system, then report this to bug-gnulib."
   ^~~~~
fseeko.c: In function 'rpl_fseeko':
fseeko.c:110:4: error: #error "Please port gnulib fseeko.c to your platform! Look at the code in fseeko.c, then report this to bug-gnulib."
   #error "Please port gnulib fseeko.c to your platform! Look at the code in fseeko.c, then report this to bug-gnulib."
    ^~~~~
`

The same issue is reported also here https://www.reddit.com/r/archlinux/comments/97gsb1/glibc_update_breaks_buildroot/ with a statement it is resolved in Buildroot 2018.08.

Can you update your repos with newer source versions? Or, at least can you provide some hints/guide how can I use newer Buildroot versions with your boards/flow?

Thanks, Michael

tholzsche, 2020-06-18:
Hi,

I am sorry this is a known issue with current linux distributions and glibc >= 2.8. There are several fixes available, have a look at:
https://github.com/buildroot/buildroot/commit/c48f8a64626c60bd1b46804b7cf1a699ff53cdf3
https://github.com/coreutils/gnulib/commit/4af4a4a71827c0bc5e0ec67af23edef4f15cee8e#diff-5bcce8ce55246264586c4aed2a45ff89

This will be fixed in the next release. Currently I recommend to use Ubuntu 18.04 LTS, 16.04 LTS or 14.04 LTS: https://releases.ubuntu.com

Regards
  Till

michael-vacek, 2020-06-18:
Thank you for the answer, I will manually patch the buildroot. Do you have an idea, when do you plan to update Enclustra xilinx-linux, xilinx-u-boot and buildroot forks to more up-to-date versions?

Thanks,
Michael