enclustra-bsp/bsp-xilinx #29: i can't make new image,what't the problem

wangyuanzhengbighead, 2020-06-23:
[image]

tholzsche, 2020-06-24:
Can you provide the complete build.log? But I assume a simple cleanup should solve these build issues.

Greets Till

wangyuanzhengbighead, 2020-06-30:
i  use "./build.sh --clear-all",  and renew ./build.sh ... also with error
this is my error log, Looks like a missing dynamic link library
[build.log](https://github.com/enclustra-bsp/bsp-xilinx/files/4849355/build.log)

tholzsche, 2020-06-30:
Which ubuntu version do you use and have you followed the installation instructions (**https://enclustra.github.io/ebe-docs/user-doc-xilinx/index_xilinx.html#ubuntu-14-04-lts**)?

wangyuanzhengbighead, 2020-07-07:
i use ubuntu-18.04-lts

tholzsche, 2020-07-07:
You need to resolve the dependencies anyway ;)

From the build.log:
_...
/home/wyz/bsp-xilinx/bin/aarch64-none-linux-gnueabi/bin/../lib/gcc/aarch64-none-linux-gnueabi/7.2.0/../../../../aarch64-none-linux-gnueabi/bin/ld: /home/wyz/bsp-xilinx/bin/aarch64-none-linux-gnueabi/bin/../libexec/gcc/aarch64-none-linux-gnueabi/7.2.0/liblto_plugin.so: error loading plugin: **libc.so.6**: cannot open shared object file: **No such file or directory**
..._

Get shared object dependencies:
_
bsp-xilinx$ ldd bin/aarch64-none-linux-gnueabi/bin/../libexec/gcc/aarch64-none-linux-gnueabi/7.2.0/liblto_plugin.so
	linux-gate.so.1 =>  (0xf77ad000)
	**libc.so.6 => /lib/i386-linux-gnu/libc.so.6 (0xf75bb000)**
	/lib/ld-linux.so.2 (0xf77ae000)
_

Have a look at the installation instructions, they should also be applicable to ubuntu-18.04-lts (https://enclustra.github.io/ebe-docs/user-doc-xilinx/index_xilinx.html#ubuntu-14-04-lts)?