enclustra-bsp/bsp-xilinx #31: Performance evaluatio of XU1

flpm2, 2020-11-06:
Hi, 
since I don't see any way to test the complete system performance using the current release v1.8.2 (i.e. no support for remote procs / unknown gpu support ). 
Could you please advise on how to obtain meaningful results?

tholzsche, 2020-11-10:
Hi, this is quite a wide field and with the EBE we provide a very flexible framework which can be extended accordingly. Nevertheless, Petalinux BSPs are currently in work, which allow a somewhat easier integration of the gpu and vcu. Furthermore there are a lot of examples for performance analysis of Xilinx, e.g.: https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/189595724/Zynq+UltraScale+MPSoC+Example+Designs

flpm2, 2020-11-11:
Hi! Thanks for the reply. In the eclustra case I don't think the petalinux bsp is applicable; the use a MSP430 to readout most of the data. Can you please suggest how to readout the available sensors from /sys instead of vivado ?

flpm2, 2020-11-11:
According to the [datasheet](https://www.xilinx.com/support/documentation/selection-guides/zynq-ultrascale-plus-product-selection-guide.pdf)of 15EG, it should be 1.5GHz; I tried to set the opp0 to this frequency, but actually it doesn't set it even though it is reported as max frequency. 

`root@mercury-xu1:~# cpufreq-info
cpufrequtils 008: cpufreq-info (C) Dominik Brodowski 2004-2009
Report errors and bugs to cpufreq@vger.kernel.org, please.
analyzing CPU 0:
  driver: cpufreq-dt
  CPUs which run at the same hardware frequency: 0 1 2 3
  CPUs which need to have their frequency coordinated by software: 0 1 2 3
  maximum transition latency: 500 us.
  hardware limits: 300 MHz - 1.50 GHz
  available frequency steps: 300 MHz, 400 MHz, 600 MHz, 1.50 GHz
  available cpufreq governors: userspace, performance
  current policy: frequency should be within 300 MHz and 1.50 GHz.
                  The governor "performance" may decide which speed to use
                  within this range.
  current CPU frequency is 1.20 GHz (asserted by call to hardware).
analyzing CPU 1:
  driver: cpufreq-dt
  CPUs which run at the same hardware frequency: 0 1 2 3
  CPUs which need to have their frequency coordinated by software: 0 1 2 3
  maximum transition latency: 500 us.
  hardware limits: 300 MHz - 1.50 GHz
  available frequency steps: 300 MHz, 400 MHz, 600 MHz, 1.50 GHz
  available cpufreq governors: userspace, performance
  current policy: frequency should be within 300 MHz and 1.50 GHz.
                  The governor "performance" may decide which speed to use
                  within this range.
  current CPU frequency is 1.20 GHz (asserted by call to hardware).
analyzing CPU 2:
  driver: cpufreq-dt
  CPUs which run at the same hardware frequency: 0 1 2 3
  CPUs which need to have their frequency coordinated by software: 0 1 2 3
  maximum transition latency: 500 us.
  hardware limits: 300 MHz - 1.50 GHz
  available frequency steps: 300 MHz, 400 MHz, 600 MHz, 1.50 GHz
  available cpufreq governors: userspace, performance
  current policy: frequency should be within 300 MHz and 1.50 GHz.
                  The governor "performance" may decide which speed to use
                  within this range.
  current CPU frequency is 1.20 GHz (asserted by call to hardware).
analyzing CPU 3:
  driver: cpufreq-dt
  CPUs which run at the same hardware frequency: 0 1 2 3
  CPUs which need to have their frequency coordinated by software: 0 1 2 3
  maximum transition latency: 500 us.
  hardware limits: 300 MHz - 1.50 GHz
  available frequency steps: 300 MHz, 400 MHz, 600 MHz, 1.50 GHz
  available cpufreq governors: userspace, performance
  current policy: frequency should be within 300 MHz and 1.50 GHz.
                  The governor "performance" may decide which speed to use
                  within this range.
  current CPU frequency is 1.20 GHz (asserted by call to hardware).

`
Can you please suggest how to set it ?

tholzsche, 2020-11-11:
It actually states up to 1.5GHz. Please check which cpu frequency is expected for your device (Table 28 in https://www.xilinx.com/support/documentation/data_sheets/ds925-zynq-ultrascale-plus.pdf)

Since you work with cpufreq-utils you may want to take a look at https://www.xilinx.com/support/answers/71070.html

flpm2, 2020-11-12:
Thank you very much! I was having the wrong expectations of my 15EG device (ME-XU1-15EG-2I-D12E).. One last thing that's bothering me is that when I set the RAM to be 4GB instead of 2GB (as suggested in [issue 10]( https://github.com/enclustra-bsp/bsp-xilinx/issues/10)). I start to get ADMA errors from the sd-card when the I/O load increases.

https://paste.ubuntu.com/p/Yw7p4hnPYV/

 Do you have any suggestions what am I doing wrong?  (currently use ubuntu 18 instead of buildroot rootfs)

tholzsche, 2020-11-12:
This is perhaps caused by a simplification of the reference design, which uses the memory size of the "smallest device". Please check the "DDR Configuration" in Vivado:
* DRAM Device Capacity: 8192
* Row Address Count: 16

flpm2, 2020-11-19:
Thanks you for the support once again. Now I'm trying to integrate analog-devices kernel 4.19 to try to use the FMC to an ADVR9009 (e.g. similar setup to zcu102 - advr9009); 

However I'm struck on some weird clock error. Do you have any ideas what might be wrong?

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 4.19.0-g3efbf7f19954 (filipm@filip-hp850g7) (gcc version 7.5.0 (Linaro GCC 7.5-2019.12)) #15 SMP Thu Nov 19 13:59:37 EET 2020
[    0.000000] Machine model: Enclustra XU1 SOM
[    0.000000] earlycon: cdns0 at MMIO 0x00000000ff000000 (options '115200n8')
[    0.000000] bootconsole [cdns0] enabled
[    0.000000] cma: Reserved 256 MiB at 0x0000000070000000
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.1
[    0.000000] percpu: Embedded 22 pages/cpu @(____ptrval____) s51992 r8192 d29928 u90112
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: enabling workaround for ARM erratum 845719
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 516096
[    0.000000] Kernel command line: console=ttycdnss0,115200 rw earlycon rootwait
[    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes)
[    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes)
[    0.000000] Memory: 1783960K/2097152K available (8830K kernel code, 570K rwdata, 3624K rodata, 704K init, 500K bss, 51048K reserved, 262144K cma-reserved)
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu:     RCU event tracing is enabled.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=4.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GIC: Adjusting CPU interface base to 0x00000000f902f000
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] arch_timer: cp15 timer(s) running at 33.33MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x7b0074340, max_idle_ns: 440795202884 ns
[    0.000003] sched_clock: 56 bits at 33MHz, resolution 30ns, wraps every 2199023255543ns
[    0.008269] Console: colour dummy device 80x25
[    0.012393] Calibrating delay loop (skipped), value calculated using timer frequency.. 66.66 BogoMIPS (lpj=133332)
[    0.022668] pid_max: default: 32768 minimum: 301
[    0.027375] Mount-cache hash table entries: 4096 (order: 3, 32768 bytes)
[    0.033918] Mountpoint-cache hash table entries: 4096 (order: 3, 32768 bytes)
[    0.041811] ASID allocator initialised with 32768 entries
[    0.046418] rcu: Hierarchical SRCU implementation.
[    0.051666] smp: Bringing up secondary CPUs ...
[    0.055929] Detected VIPT I-cache on CPU1
[    0.055960] CPU1: Booted secondary processor 0x0000000001 [0x410fd034]
[    0.056269] Detected VIPT I-cache on CPU2
[    0.056289] CPU2: Booted secondary processor 0x0000000002 [0x410fd034]
[    0.056582] Detected VIPT I-cache on CPU3
[    0.056601] CPU3: Booted secondary processor 0x0000000003 [0x410fd034]
[    0.056647] smp: Brought up 1 node, 4 CPUs
[    0.091089] SMP: Total of 4 processors activated.
[    0.095762] CPU features: detected: 32-bit EL0 Support
[    0.102494] CPU: All CPU(s) started at EL2
[    0.104941] alternatives: patching kernel code
[    0.110173] devtmpfs: initialized
[    0.117481] random: get_random_u32 called from bucket_table_alloc.isra.7+0x88/0x180 with crng_init=0
[    0.121875] Registered cp15_barrier emulation handler
[    0.126735] Registered setend emulation handler
[    0.131325] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[    0.140923] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.151990] pinctrl core: initialized pinctrl subsystem
[    0.152833] NET: Registered protocol family 16
[    0.156891] audit: initializing netlink subsys (disabled)
[    0.162080] audit: type=2000 audit(0.104:1): state=initialized audit_enabled=0 res=1
[    0.170131] vdso: 2 pages (1 code @ (____ptrval____), 1 data @ (____ptrval____))
[    0.177076] hw-breakpoint: found 6 breakpoint and 4 watchpoint registers.
[    0.184622] DMA: preallocated 256 KiB pool for atomic allocations
[    0.200672] HugeTLB registered 2.00 MiB page size, pre-allocated 0 pages
[    0.202697] vgaarb: loaded
[... 111 lines cut ...]
[    0.855477] xilinx-zynqmp-dma fd500000.dma: ignoring dependency for device, assuming no driver
[    0.859052] xilinx-zynqmp-dma fd510000.dma: ignoring dependency for device, assuming no driver
[    0.867611] xilinx-zynqmp-dma fd520000.dma: ignoring dependency for device, assuming no driver
[    0.876174] xilinx-zynqmp-dma fd530000.dma: ignoring dependency for device, assuming no driver
[    0.884742] xilinx-zynqmp-dma fd540000.dma: ignoring dependency for device, assuming no driver
[    0.893301] xilinx-zynqmp-dma fd550000.dma: ignoring dependency for device, assuming no driver
[    0.901868] xilinx-zynqmp-dma fd560000.dma: ignoring dependency for device, assuming no driver
[    0.910432] xilinx-zynqmp-dma fd570000.dma: ignoring dependency for device, assuming no driver
[... cut ...]
[    0.972817] sdhci-arasan ff170000.sdhci: ignoring dependency for device, assuming no driver
[    0.981224] xuartps ff000000.serial: ignoring dependency for device, assuming no driver
[    0.988830] zynq-gpio ff0a0000.gpio: ignoring dependency for device, assuming no driver
[    0.998694] rtc_zynqmp ffa60000.rtc: setting system clock to 1970-01-01 00:37:33 UTC (2253)
[    1.004893] of_cfs_init
[    1.007312] of_cfs_init: OK
[    1.010245] Warning: unable to open an initial console.
[    1.015409] Waiting for root device ...

Thank you once again!

tholzsche, 2020-11-20:
Hi, I would not recommend to simply use the XU1 device-tree. I would build on an existing device-tree in the analog-deivce kernel (zcu102) and just make the necessary adjustments "aliases", "gem", "sdhci1"... Hopefully this will help.

flpm2, 2020-12-01:
Managed to run latest ad linux master after I recompiled xilinx atf to latest master. Thank you for the support.

closed 2020-12-01 by flpm2