enclustra-bsp/bsp-xilinx #32: only eth0 can work? how to config it make both eth0 and eth1 work

wangyuanzhengbighead, 2021-01-04:
I use petalinux2020 to generate the kernel, my baseboard is "pe1 + xu15eg",i add some infomation in system-user.dtsi. but the eth1 also not work.what should i do?
[image]
[image]
[startlog.txt](https://github.com/enclustra-bsp/bsp-xilinx/files/5762735/startlog.txt)
[system-userdtsi.txt](https://github.com/enclustra-bsp/bsp-xilinx/files/5762736/system-userdtsi.txt)

tholzsche, 2021-01-04:
Hi. The mdio interface is shared between gem0 and gem3. The device tree should look like below. We will soon release petalinux BSPs based on 2020.1. Please contact our support for a preliminary release. Greets
```
&gem0 {
	status = "okay";
	/delete-property/ local-mac-address;
	phy-mode = "rgmii-id";
	phy-handle = <&phy0>;
	phy0: phy@3 {
		reg = <3>;

		txc-skew-ps = <1800>;
		txen-skew-ps = <420>;

		txd0-skew-ps = <420>;
		txd1-skew-ps = <420>;
		txd2-skew-ps = <420>;
		txd3-skew-ps = <420>;

		rxc-skew-ps = <900>;
		rxdv-skew-ps = <420>;

		rxd0-skew-ps = <420>;
		rxd1-skew-ps = <420>;
		rxd2-skew-ps = <420>;
		rxd3-skew-ps = <420>;		
	};
	phy1: phy@7 {
		reg = <7>;

		txc-skew-ps = <1800>;
		txen-skew-ps = <420>;

		txd0-skew-ps = <420>;
		txd1-skew-ps = <420>;
		txd2-skew-ps = <420>;
		txd3-skew-ps = <420>;

		rxc-skew-ps = <900>;
		rxdv-skew-ps = <420>;

		rxd0-skew-ps = <420>;
		rxd1-skew-ps = <420>;
		rxd2-skew-ps = <420>;
		rxd3-skew-ps = <420>;		
	};
};
```

wangyuanzhengbighead, 2021-01-05:
thanks, it works

closed 2021-01-05 by wangyuanzhengbighead