enclustra-bsp/bsp-xilinx #33: how to boot from emmc?

wangyuanzhengbighead, 2021-01-05:
I use sd card to log in to linux , copy kernel uboot and fs to emmc. i hope  to use emmc boot linux. but i do not find how to do that
i find how to set cfga and cfgb in qspi flash boot and sd card boot in 《Reference Design for Mercury+ PE1 Base Board
User Manual》

my baseboard is xu15eg + pe1

tholzsche, 2021-01-05:
Hi, please have a look at chapter "4.3  eMMC Boot" of https://raw.githubusercontent.com/enclustra/Mercury_XU1_PE1_Reference_Design/master/doc/Mercury_XU1_PE1.pdf. Greets

wangyuanzhengbighead, 2021-01-06:
thanks,The version in my hand does not have an emmc boot chapter. i can't find your version in https://www.enclustra.com/en/products/system-on-chip-modules/mercury-xu1/.

wangyuanzhengbighead, 2021-01-06:
I read the document that I need to switch DIP back and forth. 
After setting dip and usb once, start directly from emmc, no need to switch dip repeatedly.
I want to start it directly from emmc when I power it on.
can it be achieved?
if can't, What causes emmc to start, it can’t be started directly like qspi or sd card？

tholzsche, 2021-01-06:
It toggles so that the micro usb can still be used (see chapter "6.3  DIP Switches" of https://download.enclustra.com/public_files/Base_Boards/Mercury+_PE1/Mercury_PE1_User_Manual_V06.pdf). Depending on what you want, you need to make a choice. Note you can set the bootmode1 directly via IIC (see chapter "2.3.14Boot Configuration Enable Register (0x0E)" https://download.enclustra.com/public_files/Base_Boards/Mercury+_PE1/SystemController_User_Manual_PE1_V1.pdf).

closed 2021-01-11 by wangyuanzhengbighead