enclustra-bsp/bsp-xilinx #35: Toolchain issue on Dev

gbalke, 2021-04-06:
I'm using an XU9 module so I require the `develop` branch as 1.8.2 does not support the hardware. I've tried building but it appears there's an issue with your `bootgen` package?

```
INFO: Enclustra Build Environment (v1.8.2-9fdb3be)
Running under Python version 2.7.18.

Copyright (c) 2015-2021 Enclustra GmbH, Switzerland.
All rights reserved.

INFO: Fetching U-Boot
+git submodule init xilinx-uboot
+git submodule update --remote --depth 1 xilinx-uboot
+git fetch --depth 1 origin v1.8.2
+git checkout FETCH_HEAD
Target U-Boot fetched
INFO: Fetching Linux
+git submodule init xilinx-linux
+git submodule update --remote --depth 1 xilinx-linux
+git fetch --depth 1 origin v1.8.2
+git checkout FETCH_HEAD
Target Linux fetched
INFO: Fetching Buildroot
+git submodule init buildroot-rootfs
+git submodule update --remote --depth 1 buildroot-rootfs
+git fetch --depth 1 origin v1.8.2
+git checkout FETCH_HEAD
Target Buildroot fetched
+curl -O http://www.enclustra.com/binaries/enclustra-bsp/v1.8.2/arm-none-linux-gnueabi-static-i686.tar.bz2
+curl -O http://www.enclustra.com/binaries/enclustra-bsp/v1.8.2/aarch64-none-linux-gnueabi-static-i686.tar.gz
+curl -O http://www.enclustra.com/binaries/enclustra-bsp/v1.8.2/uboot-tools-i686-static.tar.gz
+curl -O http://www.enclustra.com/binaries/enclustra-bsp/v1.8.2/device-tree-compiler-i686-static.tar.gz
+curl -O http://www.enclustra.com/binaries/enclustra-bsp/v1.8.2/bootgen_2020_1.tar.gz
ERROR: Error while unpacking bootgen_2020_1.tar.gz toolchain. file could not be opened successfully - deleting.
ERROR: Failed to acquire toolchain, skipping build! Required toolchains: arm-none-linux-gnueabi, aarch64-none-linux-gnueabi, uboot-tools, device-tree-compiler, bootgen, boot_config, atf
INFO: --------------------------------------------------------------------------------
ERROR: BUILD FAILED
INFO: Device: Zynq-Ultrascale+ Mercury XU9 Mercury ST1 MMC
INFO: Targets: U-Boot (fetch + build), Buildroot (fetch + build), Linux (fetch + build)
INFO: Binaries: ME-XU9-4CG-1E-D11E
```

hammerhg, 2021-05-31:
Same issue when building for `ST1` board from `develop` branch

```INFO: Fetching U-Boot
+git submodule init xilinx-uboot
+git submodule update --remote --depth 1 xilinx-uboot
+git fetch --depth 1 origin v1.8.2
+git checkout FETCH_HEAD
Target U-Boot fetched
INFO: Fetching Linux
+git submodule init xilinx-linux
+git submodule update --remote --depth 1 xilinx-linux
+git fetch --depth 1 origin v1.8.2
+git checkout FETCH_HEAD
Target Linux fetched
INFO: Fetching Buildroot
+git submodule init buildroot-rootfs
+git submodule update --remote --depth 1 buildroot-rootfs
+git fetch --depth 1 origin v1.8.2
+git checkout FETCH_HEAD
Target Buildroot fetched
+curl -O http://www.enclustra.com/binaries/enclustra-bsp/v1.8.2/bootgen_2020_1.tar.gz
ERROR: Error while unpacking bootgen_2020_1.tar.gz toolchain. file could not be opened successfully - deleting.
ERROR: Failed to acquire toolchain, skipping build! Required toolchains: arm-none-linux-gnueabi, aarch64-none-linux-gnueabi, uboot-tools, device-tree-compiler, bootgen, boot_config, atf
INFO: --------------------------------------------------------------------------------
ERROR: BUILD FAILED
INFO: Device: Zynq-Ultrascale+ Mercury XU1 Mercury ST1 QSPI 
INFO: Targets: U-Boot (fetch + build), Buildroot (fetch + build), Linux (fetch + build) 
INFO: Binaries: ME-XU1-6CG-1E-D11E
```

gbalke, 2021-05-31:
@hammerhg I ended up just using petalinux. Unless you get an update soon, I'd recommend checking out that build process (definitely more work but at least it works).

abuerkler, 2021-06-01:
Dear gbalke / Dear hammerhg

Please have a look at the documentation: https://github.com/enclustra-bsp/bsp-xilinx/blob/develop/documentation/2_GUI.md
It looks like you haven't set EBE_RELEASE variable (export EBE_RELEASE=update).

We are close to a new release and the develop branch will be merged into master branch soon. Setting the mentioned variable will not be required in master branch.

Best regards
Andreas

hammerhg, 2021-06-09:
The issue is now solved.

Having several failed buils it was required to pull the source from scratch and build from `develop` branch having `export EBE_RELEASE=update`

gbalke, 2021-06-09:
Thanks for testing that out @hammerhg! I would be interested in using it but I've already set up my petalinux workflow so it might be more effort than it's worth for me haha. Hope it's a good workflow for you!

closed 2021-06-09 by gbalke