enclustra-bsp/bsp-xilinx #36: Problem booting using SD Card

GNiramay, 2021-07-09:
Dear all,

I am a physics grad student with very little experience in electronics and I'm having trouble booting my zynq ultrascale+ board using an SD Card. Can someone please help me?

When I switched on the board, 3 LEDs with labels PWGD,SC, E started glowing. I believe that the LED with label E corresponds to some error in booting. I tried to modify the SD Card in different ways but always ended up with this LED glowing.

I followed the steps described in the [Enclustra git repository](https://github.com/enclustra-bsp/bsp-xilinx.git) to prepare all the necessary files (boot.bin, uboot.scr etc.). Then I transferred them to the SD card as described on the [Xilinx confluence page](https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/18841655/Prepare+Boot+Medium#PrepareBootMedium-SDBoot). I also copied additional files to the boot partition of the SD Card as described in [Enclustra documentation](https://github.com/enclustra-bsp/bsp-xilinx/blob/master/documentation/4_Deployment.md#sd-card-mmc).

Then, before switching the board on, I made the following changes on the board.

1. Set the power switch to the left.
2. Connect VSel A to B and VSel B to B

Please let me know if you need any other information.

Hardware information:

zynq mezzanine: ME-XU5-4EV-1I-D11E-G1-R1.2
baseboard: ME-PE1-300-W-R4.6 

SD Card - related information :

```
dmesg | grep mmc
[    1.550127] sr 1:0:0:0: [sr0] scsi3-mmc drive: 24x/24x writer dvd-ram cd/rw xa/form2 cdda tray
[  747.602352] mmc0: new ultra high speed SDR50 SDHC card at address aaaa
[  747.705676] mmcblk0: mmc0:aaaa SC32G 29.7 GiB 
[  747.723973]  mmcblk0: p1 p2
[  748.281463] EXT4-fs (mmcblk0p2): mounted filesystem with ordered data mode. Opts: (null)
```
```
sudo fdisk -l /dev/mmcblk0
Disk /dev/mmcblk0: 29.74 GiB, 31914983424 bytes, 62333952 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xc86ba88b

Device         Boot  Start      End  Sectors  Size Id Type
/dev/mmcblk0p1 *      2048   411647   409600  200M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      411648 62333951 61922304 29.5G 83 Linux
lsblk | grep mmc
mmcblk0     179:0    0  29.7G  0 disk 
├─mmcblk0p1 179:1    0   200M  0 part /mnt/boot
└─mmcblk0p2 179:2    0  29.5G  0 part /mnt/rootfs
```
```
ll /mnt/boot/
total 39M
drwxr-xr-x 2 niramay niramay  512 Dec 31  1969 .
drwxr-xr-x 5 root    root    4.0K Jun 23 17:32 ..
-rw-r--r-- 1 niramay niramay 6.5M Jun 23 18:12 boot.bin
-rw-r--r-- 1 niramay niramay  28K Jun 24 14:23 devicetree.dtb
-rw-r--r-- 1 niramay niramay  17M Jun 24 14:23 Image
-rw-r--r-- 1 niramay niramay  17M Jun 24 14:23 Image.bin
-rw-r--r-- 1 niramay niramay  354 Jun 24 14:24 uboot_ramdisk.scr
-rw-r--r-- 1 niramay niramay  364 Jun 24 14:24 uboot.scr

```
```
ll /mnt/rootfs/
total 68K
drwxr-xr-x 17 root root 4.0K Jun 19 12:29 .
drwxr-xr-x  5 root root 4.0K Jun 23 17:32 ..
drwxr-xr-x  2 root root 4.0K Jun 21 15:45 bin
drwxr-xr-x  4 root root 4.0K Jun 19 09:53 dev
drwxr-xr-x  8 root root 4.0K Jun 21 15:46 etc
drwxr-xr-x  4 root root 4.0K Jun 21 15:45 lib
lrwxrwxrwx  1 root root    3 Jun 19 12:06 lib64 -> lib
lrwxrwxrwx  1 root root   11 Jun 19 12:29 linuxrc -> bin/busybox
drwxr-xr-x  2 root root 4.0K Jun 19 09:53 media
drwxr-xr-x  2 root root 4.0K Jun 19 09:53 mnt
drwxr-xr-x  2 root root 4.0K Jun 19 09:53 opt
drwxr-xr-x  2 root root 4.0K Jun 19 09:53 proc
drwx------  2 root root 4.0K Jun 19 09:53 root
drwxr-xr-x  2 root root 4.0K Jun 19 09:53 run
drwxr-xr-x  2 root root 4.0K Jun 21 15:45 sbin
drwxr-xr-x  2 root root 4.0K Jun 19 09:53 sys
drwxrwxrwt  2 root root 4.0K Jun 19 09:53 tmp
drwxr-xr-x  7 root root 4.0K Jun 21 15:46 usr
drwxr-xr-x  6 root root 4.0K Jun 21 15:46 var
```

abuerkler, 2021-07-09:
Hi Niramay

The LED labelled with E is active when the boot process fails.
How did you configure the DIP switches of the Mercury PE1 base board?  (CFG A = "0001", CFG B = "0000" ?)
Did you select the ME-XU5-4EV-1I-D11E-G1-R1.2 variant when you created the build?

Best regards,
Andreas

GNiramay, 2021-07-13:
Dear Andreas,

Sorry for the late reply. I was caught up with a conference.

DIp switch configuration: *CFG A = 0000, CFG B = 0000*

I created the build with the following commands. (I am using Ubuntu 20.04)

*git clone https://github.com/enclustra-bsp/bsp-xilinx.git
<https://github.com/enclustra-bsp/bsp-xilinx.git>*
*./build.sh -d Zynq-Ultrascale+/Mercury_XU5/Mercury_PE1/MMC*

Please let me know if I did something wrong.

Thank you and regards,
Niramay

abuerkler, 2021-07-15:
Hi Niramay

If you start the build with:
./build.sh -d Zynq-Ultrascale+/Mercury_XU5/Mercury_PE1/MMC

The ME-XU5-2CG-1E-D10H variant is chosen as default one. Please try to specify the module variant with:
./build.sh -d Zynq-Ultrascale+/Mercury_XU5/Mercury_PE1/MMC -o 3

The variant index can be determined by adding '--list-dev-options':

./build.sh -d Zynq-Ultrascale+/Mercury_XU5/Mercury_PE1/MMC --list-dev-options
Available options for Zynq-Ultrascale+/Mercury_XU5/Mercury_PE1/MMC:
1. ME-XU5-3EG-2I-D11E
2. ME-XU5-2CG-1E-D10H (default)
3. ME-XU5-4EV-1I-D11E-G1
4. ME-XU5-4CG-1E-D11E-G1
5. ME-XU5-5EV-2I-D12E-G1
6. ME-XU5-5EV-3E-D13E
7. ME-XU5-4EV-1I-D11E
8. ME-XU5-2EG-1I-D11E
9. ME-XU5-5EV-1E-D11E
10. ME-XU5-5EV-2I-D12E

Best regards,
Andreas

GNiramay, 2021-07-15:
Dear all,

Thank you for your valuable information. I followed the build as you
described and it seems to have worked. My only concern now is that one of
the messages that were displayed on the screen were: "*[    4.856970]
platform regulatory.0: Direct firmware load for regulatory.db failed with
error -2*". Can you tell me how to fix it?

The following are the commands I followed after creating the boot image.

./build.sh -d Zynq-Ultrascale+/Mercury_XU5/Mercury_PE1/MMC -o 3

Then keeping the partition structure constant, the files were transferred
to the SD card in the following way.

cd bsp-xilinx/out_Zynq-Ultrascale+_Mercury_XU5_Mercury_PE1_MMC

sudo mount /dev/mmcblk0p1 /mnt/boot/

sudo mount /dev/mmcblk0p2 /mnt/rootfs/

sudo rm -f /mnt/boot/*

sudo rm -rf /mnt/rootfs/*

cp boot.bin /mnt/boot/

cp devicetree.dtb /mnt/boot/

cp uboot.scr /mnt/boot/

cp Image /mnt/boot/

sudo tar -xvf rootfs.tar -C /mnt/rootfs/

sudo umount /mnt/boot

sudo umount /mnt/rootfs

Circuit connections:

CFG A = 0000

CFG B = 0000

VSEL A = B

VSEL B = B

[

After powering on the board and pressing the button "PWR", the LED's with
the following labels started glowing:

On the board: PWGD, DONE, SC

On the Zynq chip: 1

]

Connect the USB cable to the your laptop (the host)

In the host terminal, enter the following command to open a serial terminal
to talk to our board.

sudo screen /dev/ttyUSB1 115200

buildroot login: root

Password: root
[... 350 lines cut ...]
[    4.965522] Run /sbin/init as init process
[    5.076873] EXT4-fs (mmcblk1p2): re-mounted. Opts: (null)
Starting syslogd: OK
Starting klogd: OK
Running sysctl: OK
Initializing random number generator: OK
Saving random seed: [    5.139552] random: dd: uninitialized urandom read (512 bytes read)
OK
Starting network: OK
Starting dhcpcd...
no interfaces have a carrier
forked to background, child pid 200
[    5.241360] pps pps0: new PPS source ptp0
[    5.247757] macb ff0b0000.ethernet: gem-ptp-timer ptp clock registered.
[    5.256933] pps pps1: new PPS source ptp1
[    5.260967] macb ff0c0000.ethernet: gem-ptp-timer ptp clock registered.
[    5.290371] random: dhcpcd: uninitialized urandom read (112 bytes read)
[    6.947696] random: crng init done
Starting sshd: OK
Starting tcf-agent: OK

Welcome to Buildroot
buildroot login: root
Password:
#

abuerkler, 2021-07-20:
Hi Niramay

The error message about the missing regulatory.db file comes from wireless support. If you don't need WIFI, you can disable it in the kernel configuration (CFG_CFG80211). The easiest way is to remove the corresponding lines (everything containing "80211") from the file sources/xilinx-linux/arch/arm64/configs/xilinx_zynqmp_defconfig.

 Best regards,
 Andreas

GNiramay, 2021-07-22:
@abuerkler, thanks a lot for answering all of my questions. With this, I think I can close the issue

closed 2021-07-22 by GNiramay