enclustra-bsp/bsp-xilinx #41: ERROR: Error running make EBE_OVERLAYS={ebe_overlays} -j3 for Buildroot

DanW0830, 2023-04-27:
wget --passive-ftp -nd -t 3 -O '/home/dwinsor/workspace/bsp-xilinx/bsp-xilinx/sources/buildroot-rootfs/output/build/.u-boot-2021.07.tar.bz2.DuSKRz/output' 'http://www.enclustra.com/binaries/enclustra-bsp/busybox-rootfs/u-boot-2021.07.tar.bz2' 
--2023-04-26 16:51:12--  http://www.enclustra.com/binaries/enclustra-bsp/busybox-rootfs/u-boot-2021.07.tar.bz2
Resolving www.enclustra.com (www.enclustra.com)... 172.67.170.27, 104.21.28.2, 2606:4700:3033::ac43:aa1b, ...
Connecting to www.enclustra.com (www.enclustra.com)|172.67.170.27|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2023-04-26 16:51:13 ERROR 404: Not Found.

What can I do to fix this?

abuerkler, 2023-04-28:
Hi DanW0830

Can you please provide more information what you did to get to this error. Thank you.

Best regards,
Andreas

DanW0830, 2023-04-28:
Our company doesn't allow FTP connections.
Found that there are parts of the build which use ftp.   
Changing them to use https:  fixed the issue.

DanW0830, 2023-04-28:
Here's a patch that fixes it for 1.11

diff --git a/package/mmc-utils/mmc-utils.mk b/package/mmc-utils/mmc-utils.mk
index c8473786..58fa38e5 100644
--- a/package/mmc-utils/mmc-utils.mk
+++ b/package/mmc-utils/mmc-utils.mk
@@ -5,7 +5,7 @@
 ################################################################################
 
 MMC_UTILS_VERSION = d40ec535b9d4e4c974e8c2fbfb422cd0348cc5e8
-MMC_UTILS_SITE = https://git.kernel.org/pub/scm/linux/kernel/git/cjb/mmc-utils.git
+MMC_UTILS_SITE =  http://sources.buildroot.net/mmc-utils
 MMC_UTILS_SITE_METHOD = git
 MMC_UTILS_LICENSE = GPL-2.0
 MMC_UTILS_LICENSE_FILES = mmc.h
diff --git a/package/mtd/mtd.mk b/package/mtd/mtd.mk
index 45e7c0c6..61328be9 100644
--- a/package/mtd/mtd.mk
+++ b/package/mtd/mtd.mk
@@ -6,7 +6,7 @@
 
 MTD_VERSION = 2.1.3
 MTD_SOURCE = mtd-utils-$(MTD_VERSION).tar.bz2
-MTD_SITE = ftp://ftp.infradead.org/pub/mtd-utils
+MTD_SITE = https://infraroot.at/pub/mtd
 MTD_LICENSE = GPL-2.0
 MTD_LICENSE_FILES = COPYING
 MTD_CPE_ID_VENDOR = mtd-utils_project
diff --git a/package/uboot-tools/uboot-tools.mk b/package/uboot-tools/uboot-tools.mk
index e48a4f6c..29a19c65 100644
--- a/package/uboot-tools/uboot-tools.mk
+++ b/package/uboot-tools/uboot-tools.mk
@@ -6,7 +6,7 @@
 
 UBOOT_TOOLS_VERSION = 2021.07
 UBOOT_TOOLS_SOURCE = u-boot-$(UBOOT_TOOLS_VERSION).tar.bz2
-UBOOT_TOOLS_SITE = ftp://ftp.denx.de/pub/u-boot
+UBOOT_TOOLS_SITE = https://ftp.denx.de/pub/u-boot
 UBOOT_TOOLS_LICENSE = GPL-2.0+
 UBOOT_TOOLS_LICENSE_FILES = Licenses/gpl-2.0.txt
 UBOOT_TOOLS_CPE_ID_VENDOR = denx