enclustra-bsp/bsp-xilinx #44: remoteproc0 not available after loading the zynqmp_r5_remoteproc module

bitscribble, 2024-07-23:
I ran into a problem activating OpenAMP on an Enclustra Andromeda XZU65.
Even if i do a modprobe of the zynqmp_r5_remoteproc module, no R5 core seems to be available.
There is no remoteproc0 device available under /sys/class/remoteproc.

My build configuration:

`Device:`
`Zynq-Ultrascale+ Andromeda XZU65 Mercury ST1 MMC`

`Targets:`
`U-Boot (fetch + build)`
`Buildroot (fetch + build)`
`Linux (fetch + build)`

`Binaries:`
`AM-XZU65-7EV-2I-D3E`

Am i missing something?

abuerkler, 2024-07-25:
I'm not familiar with OpenAMP and we haven't tested that with our Buildroot Linux. As far as I can see, the remoteproc kernel module expects an entry in the devicetree (See following documentation: https://github.com/Xilinx/linux-xlnx/blob/master/Documentation/devicetree/bindings/remoteproc/xilinx%2Czynqmp-r5-remoteproc.yaml). This entry is not present in any of our devicetree files and needs to be added manually.

bitscribble, 2024-07-26:
Thank you for this useful hint. I will give it a try.