enclustra-bsp/bsp-xilinx #46: Ethernet problems on PE1+XU8

sombresleep, 2024-12-04:
I'm trying to connect my PE1+XU8, but for some reason there's no valid connection.

I build Linux with both BSP-Linux and PetaLinux using Enclustra PetaLinux documentation. The result is exactly the same in both cases. For the purity of the experiment I didn't change any default setting (device tree, kernel options, etc. Just a default build).

When Linux boots I see that it sees the Ethernet connection, but fails to get an IP and assigns a non-valid IP to itself in 169.254.X.X range using IPv4LL. I can't ping the device and the device can't ping anything else in the network.

It's not the DHCP server's issue because when I connect the same Ethernet cable to any other device (Raspberry Pi), it receives an IP without any problem.

Here's some logs and data:

**ifconfig**
```
ifconfig
eth0      Link encap:Ethernet  HWaddr 20:B0:F7:09:D4:AE
          inet addr:169.254.57.209  Bcast:169.254.255.255  Mask:255.255.0.0
          inet6 addr: fe80::da90:9521:2260:c0d6/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:35 errors:0 dropped:0 overruns:0 frame:0
          TX packets:55 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:3184 (3.1 KiB)  TX bytes:13588 (13.2 KiB)
          Interrupt:29

eth1      Link encap:Ethernet  HWaddr 20:B0:F7:09:D4:AF
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:30

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
```

**dmesg | grep eth** 
```
[    0.000000] psci: probing for conduit method from DT.
[    2.225819] usbcore: registered new interface driver cdc_ether
[    4.226920] macb ff0b0000.ethernet: Not enabling partial store and forward
[    4.235417] macb ff0b0000.ethernet eth0: Cadence GEM rev 0x50070106 at 0xff0b0000 irq 29 (20:b0:f7:09:d4:ae)
[    4.245533] macb ff0e0000.ethernet: Not enabling partial store and forward
[    4.260176] macb ff0e0000.ethernet eth1: Cadence GEM rev 0x50070106 at 0xff0e0000 irq 30 (20:b0:f7:09:d4:af)
[    5.291460] macb ff0b0000.ethernet eth0: PHY [ff0b0000.ethernet-ffffffff:03] driver [Micrel KSZ9031 Gigabit PHY] (irq=POLL)
[    5.302611] macb ff0b0000.ethernet eth0: configuring for phy/rgmii-id link mode
[    5.316616] macb ff0b0000.ethernet: gem-ptp-timer ptp clock registered.
[    5.387605] macb ff0e0000.ethernet eth1: PHY [ff0b0000.ethernet-ffffffff:07] driver [Micrel KSZ9031 Gigabit PHY] (irq=POLL)
[    5.398748] macb ff0e0000.ethernet eth1: configuring for phy/rgmii-id link mode
[    5.412753] macb ff0e0000.ethernet: gem-ptp-timer ptp clock registered.
[    9.413106] macb ff0b0000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[    9.420792] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
```

**cat /var/log/messages | grep -Ei 'dhcp'**
```
Jan  1 00:00:08 buildroot daemon.info dhcpcd[211]: dhcpcd-9.4.1 starting
Jan  1 00:00:08 buildroot daemon.info dhcpcd[214]: DUID 00:01:00:01:c7:92:bc:88:20:b0:f7:09:d4:af
Jan  1 00:00:08 buildroot daemon.warn dhcpcd[214]: no interfaces have a carrier
Jan  1 00:00:08 buildroot daemon.info dhcpcd[214]: eth0: waiting for carrier
Jan  1 00:00:08 buildroot daemon.info dhcpcd[214]: eth1: waiting for carrier
Jan  1 00:00:12 buildroot daemon.info dhcpcd[214]: eth0: carrier acquired
Jan  1 00:00:12 buildroot daemon.info dhcpcd[214]: eth0: IAID f7:09:d4:ae
Jan  1 00:00:12 buildroot daemon.info dhcpcd[214]: eth0: adding address fe80::da90:9521:2260:c0d6
Jan  1 00:00:12 buildroot daemon.info dhcpcd[214]: eth0: soliciting an IPv6 router
Jan  1 00:00:13 buildroot daemon.info dhcpcd[214]: eth0: soliciting a DHCP lease
Jan  1 00:00:18 buildroot daemon.info dhcpcd[214]: eth0: probing for an IPv4LL address
Jan  1 00:00:24 buildroot daemon.info dhcpcd[214]: eth0: using IPv4LL address 169.254.57.209
Jan  1 00:00:24 buildroot daemon.info dhcpcd[214]: eth0: adding route to 169.254.0.0/16
Jan  1 00:00:24 buildroot daemon.info dhcpcd[214]: eth0: adding default route
Jan  1 00:00:26 buildroot daemon.warn dhcpcd[214]: eth0: no IPv6 Routers available
Jan  1 00:12:20 buildroot daemon.info dhcpcd[214]: control command: dhcpcd
Jan  1 00:12:20 buildroot daemon.err dhcpcd[214]: control_free: No such file or directory
Jan  1 00:12:20 buildroot daemon.info dhcpcd[214]: lo: IAID 00:00:00:00
Jan  1 00:12:20 buildroot daemon.info dhcpcd[214]: lo: adding address fe80::dd04:4ae2:33d6:101d
Jan  1 00:12:20 buildroot daemon.info dhcpcd[214]: lo: using static address 127.0.0.1/8
Jan  1 00:12:20 buildroot daemon.info dhcpcd[214]: lo: adding route to 127.0.0.0/8
Jan  1 00:12:20 buildroot daemon.info dhcpcd[214]: sit0: waiting for carrier
Jan  1 00:12:20 buildroot daemon.info dhcpcd[214]: sit0: carrier acquired
Jan  1 00:12:20 buildroot daemon.info dhcpcd[214]: sit0: IAID 00:00:00:00
Jan  1 00:12:20 buildroot daemon.err dhcpcd[214]: sit0: IAID conflicts with one assigned to lo
Jan  1 00:12:20 buildroot daemon.info dhcpcd[214]: sit0: adding address fe80::9e1b:ce4:7cbc:6c97
Jan  1 00:12:20 buildroot daemon.info dhcpcd[214]: sit0: soliciting an IPv6 router
Jan  1 00:12:21 buildroot daemon.info dhcpcd[214]: sit0: soliciting a DHCP lease
Jan  1 00:12:21 buildroot daemon.err dhcpcd[288]: ps_bpf_start_bpf: bpf_open: Invalid argument
Jan  1 00:12:21 buildroot daemon.err dhcpcd[288]: ps_root_recvmsg: Invalid argument
Jan  1 00:12:25 buildroot daemon.err dhcpcd[214]: ps_sendpsmmsg: Connection refused
Jan  1 00:12:25 buildroot daemon.err dhcpcd[214]: ps_root_recvmsgcb: failed to send message to pid 288: Connection refused
Jan  1 00:12:26 buildroot daemon.info dhcpcd[214]: sit0: probing for an IPv4LL address
Jan  1 00:12:26 buildroot daemon.err dhcpcd[289]: ps_bpf_start_bpf: bpf_open: Invalid argument
Jan  1 00:12:26 buildroot daemon.err dhcpcd[289]: ps_root_recvmsg: Invalid argument
Jan  1 00:12:28 buildroot daemon.err dhcpcd[214]: ps_sendpsmmsg: Connection refused
Jan  1 00:12:28 buildroot daemon.err dhcpcd[214]: ps_root_recvmsgcb: failed to send message to pid 289: Connection refused
Jan  1 00:12:30 buildroot daemon.err dhcpcd[214]: ps_root_recvmsg: Invalid argument
Jan  1 00:12:32 buildroot daemon.info dhcpcd[214]: sit0: using IPv4LL address 169.254.185.103
Jan  1 00:12:32 buildroot daemon.info dhcpcd[214]: sit0: adding route to 169.254.0.0/16
Jan  1 00:12:32 buildroot daemon.info dhcpcd[214]: sit0: adding default route
Jan  1 00:12:32 buildroot daemon.info dhcpcd[214]: eth0: deleting default route
Jan  1 00:12:32 buildroot daemon.warn dhcpcd[214]: sit0: no IPv6 Routers available
Jan  1 00:12:33 buildroot daemon.err dhcpcd[298]: ps_bpf_start_bpf: bpf_open: Invalid argument
Jan  1 00:12:33 buildroot daemon.err dhcpcd[298]: ps_root_recvmsg: Invalid argument
Jan  1 00:12:48 buildroot daemon.err dhcpcd[214]: ps_sendpsmmsg: Connection refused
Jan  1 00:12:48 buildroot daemon.err dhcpcd[214]: ps_root_recvmsgcb: failed to send message to pid 298: Connection refused
Jan  1 00:13:20 buildroot daemon.err dhcpcd[303]: ps_bpf_start_bpf: bpf_open: Invalid argument
Jan  1 00:13:20 buildroot daemon.err dhcpcd[303]: ps_root_recvmsg: Invalid argument
```
Also there's absolutely nothing in `lsmod`

Am I missing something? I'm completely stuck with this.

andreas-buerkler-enclustra, 2024-12-04:
I did a quick test on my hardware, but I can't reproduce your issue. I built a Linux design for XU8 and PE1 from master branch without any modifications and I get an address for ETH0 from the DHCP server:

\# cat /var/log/messages | grep -Ei 'dhcp'
Jan  1 00:01:15 buildroot daemon.info dhcpcd[195]: dhcpcd-9.4.1 starting
Jan  1 00:01:15 buildroot daemon.info dhcpcd[198]: DUID 00:01:00:01:c7:92:bc:87:20:b0:f7:03:c4:81
Jan  1 00:01:15 buildroot daemon.warn dhcpcd[198]: no interfaces have a carrier
Jan  1 00:01:15 buildroot daemon.info dhcpcd[198]: eth0: waiting for carrier
Jan  1 00:01:15 buildroot daemon.info dhcpcd[198]: eth1: waiting for carrier
Jan  1 00:01:19 buildroot daemon.info dhcpcd[198]: eth0: carrier acquired
Jan  1 00:01:19 buildroot daemon.info dhcpcd[198]: eth0: IAID f7:03:c4:80
Jan  1 00:01:19 buildroot daemon.info dhcpcd[198]: eth0: adding address fe80::4667:511d:24b3:38f2
Jan  1 00:01:19 buildroot daemon.info dhcpcd[198]: eth0: soliciting an IPv6 router
Jan  1 00:01:21 buildroot daemon.err dhcpcd[198]: eth0: read_lease: truncated lease
Jan  1 00:01:21 buildroot daemon.info dhcpcd[198]: eth0: soliciting a DHCP lease
Jan  1 00:01:24 buildroot daemon.info dhcpcd[198]: eth0: offered 192.168.2.101 from 192.168.2.1
Jan  1 00:01:24 buildroot daemon.info dhcpcd[198]: eth0: probing address 192.168.2.101/24
Jan  1 00:01:30 buildroot daemon.info dhcpcd[198]: eth0: leased 192.168.2.101 for 180000 seconds
Jan  1 00:01:30 buildroot daemon.info dhcpcd[198]: eth0: adding route to 192.168.2.0/24
Jan  1 00:01:30 buildroot daemon.info dhcpcd[198]: eth0: adding default route via 192.168.2.1`

I was using Tftpd64 running on Windows 10 as DHCP server for this test.

sombresleep, 2024-12-05:
Well... It works now.

Basically my previous setup involved my laptop sharing the wifi connection through the ethernet cable. It worked perfectly with Raspberi Pi. But not with XU8 for some reason. 

I connected it to the work network directly and it got its ip address without any problems. I can't explain how this can be different.

closed 2024-12-10 by sombresleep