enclustra-bsp/bsp-xilinx #47: Problems with editing MIO in BSP

sombresleep, 2024-12-12:
I'm trying to enable CAN0 in the PetaLinux project based on `Petalinux_ME-XU8-7EV-2I-D12E_PE1_SD` and no luck so far.

How do I modify PS peripherals in the BSP for PetaLinux correctly?

Your PetaLinux documentations says (https://github.com/enclustra/PetalinuxDocumentation/blob/master/doc/BSP.md) `The provided xsa [...] Make your changes to the hardware in Vivado.` 

But I can't open that XSA file in Vivado. When I try opening it, it says `
open hw platform is not supported for platforms in which the hardware/programmable logic cannot be modified.  The Hardware Platform can be used for application software development in the Vitis Tool.`

I decided to take another route and to modify the Reference Design XPR and then generate the XSA file. But when I run "petalinux-config --get-hw-description=<path>.xsa" it fails with: ERROR: Failed to generate /home/art/enc-peta/ME-XU8-7EV-2I-D12E PE1 SD/build/misc/config/Kconfig.syshw ERROR: Failed to Kconfig project ERROR: Failed to generate System hardware Kconfig file. The config log says "error "No memory base address and high address is provided" What am I missing? 

I tried another approach and edited psu_init.c, psu_init_gpl.c and psu_init.tcl and enable CAN and assign it to certain pins there. (obviously, I also enabled can0 in the device tree). But still no luck. Dmesg always says

```
[    3.922220] zynqmp_gpd_attach_dev() domain5 request failed for node 47: -13
[    3.929180] xilinx_can ff060000.can: failed to add to PM domain domain5: -13
[    3.936222] xilinx_can: probe of ff060000.can failed with error -13
```

What's the correct way of doing it? Is there a chance of getting an XPR project for ME-XU8-7EV-2I-D12E_PE1_SD?

andreas-buerkler-enclustra, 2024-12-16:
The Petalinux BSP was generated out of the xsa file, exported from the Vivado reference design. This reference design can be found in github, e.g. https://github.com/enclustra/Mercury_XU8_PE1_Reference_Design/tree/2022.1_v1.1.2.

To enable CAN0 peripheral, you need to create this Vivado project, modify the PS configuration to enable CAN0, build the design, export the xsa file and reconfigure the Petalinux project (petalinux-config --get-hw-description=<path-to-exported-xsa-file>). When you export the xsa file, it is important to enable the option to include the bitstream in the xsa file.

sombresleep, 2024-12-16:
Thank you for your response! I've already tried all those steps. Here they are in detail:

1. Clone [Mercury_XU8_PE1_Reference_Design, check out 2022.1_v1.1.2
2. Change my module in settings.tcl to `ME-XU8-7EV-2I-D12E`
3. Run `source scripts/create_project.tcl` in Vivado.
4. Costomize Zynq UltraScale+ MPCSoC > I/O Configuration > Enable CAN0 > Assign MIO 42 .. 43
5. Save Block Diagram
6. Generate Bit stream
7. Export Hardware
8. petalinux-config --get-hw-description=path_to_xsa.xsa

The result is always this:

```
[INFO] Sourcing buildtools
INFO: Getting hardware description...
INFO: Renaming Mercury_XU8_PE1.xsa to system.xsa
[INFO] Generating Kconfig for project
ERROR: Failed to generate /home/art/enc-peta/ME-XU8-7EV-2I-D12E_PE1_SD/build/misc/config/Kconfig.syshw
ERROR: Failed to Kconfig project
ERROR: Failed to generate System hardware Kconfig file.
```

Build.log:

```
[INFO] Sourcing buildtools
[INFO] Generating Kconfig for project
INFO: [Hsi 55-2053] elapsed time for repository (/home/art/peta2/tools/xsct/data/embeddedsw) loading 0 seconds
No memory base address and high address is provided
    while executing
"error "No memory base address and high address is provided""
    (procedure "plnx_gen_memory_bank_kconfig" line 7)
    invoked from within
"plnx_gen_memory_bank_kconfig "" ${bankbaseaddr} ${bankhighaddr} "${name}" "${memorykconfprefix}""
    ("foreach" body line 13)
    invoked from within
"foreach hd ${hds} {
			set name [hsi get_property NAME ${hd}]
			if {[lsearch ${cpuslaves} ${name}] < 0} {
				continue
			}
			set kname [plnx_fix_kc..."
    ("foreach" body line 25)
    invoked from within
"foreach m ${mapping} {
		set ipname [lindex ${m} 0]
		set devinfo [get_ip_device_info memory ${m}]
		set has_bank [lindex [get_ip_property_info has_ba..."
    (procedure "plnx_gen_conf_memory" line 14)
    invoked from within
"${pproc} ${elements} "${pkconfprefix}" "${cpuname}" "${cpuslaves}""
    ("foreach" body line 11)
    invoked from within
"foreach m [lreplace ${mapping} 0 0] {
			set class [lindex ${m} 0]
			set classcpuipnames [get_ip_property_info "processor_ip" ${m}]
			if { [llength ..."
    ("foreach" body line 13)
    invoked from within
"foreach c [lreplace ${retcpus} 0 0] {
		set cpuname [lindex ${c} 0]
		set cpuarch [lindex [get_ip_property_info "arch" ${c}] 0]
		set cpukname ${cpuna..."
    (procedure "plnx_gen_hwsysconf" line 52)
    invoked from within
"${tclproc} ${cmdline}"
    invoked from within
"if { "[info procs ${tclproc}]" eq "${tclproc}"} {
	${tclproc} ${cmdline}
} else {
	error "proc ${tclproc} doesn't exit."
}"
    (file "/home/art/enc-peta/ME-XU8-7EV-2I-D12E_PE1_SD/build/misc/config/hw-description/hw-description.tcl" line 1673)
ERROR: Failed to generate /home/art/enc-peta/ME-XU8-7EV-2I-D12E_PE1_SD/build/misc/config/Kconfig.syshw
ERROR: Failed to Kconfig project
```

I tried not modifying the project at all. Just create_project.tcl > Generate bitstream > Export > petalinux-config

It's always this `No memory base address and high address is provided` error.

andreas-buerkler-enclustra, 2024-12-17:
To check if this error is related to your xsa file, you could execute "petalinux-config --get-hw-description" with the prebuilt xsa file of the reference design (https://github.com/enclustra/Mercury_XU8_PE1_Reference_Design/releases/download/2022.1_v1.1.2/ReferenceDesign_ME-XU8-7EV-2I-D12E_PE1.zip).

Did you already use the "petalinux-config --get-hw-description" command successfully in another project with the same Petalinux installation? Or is this the first project? It could be that this error is related to your Petalinux installation and maybe a dependency is missing.

E.g. for Ubuntu 22.04, following dependencies are required:
```

sudo dpkg --add-architecture i386
sudo apt update
sudo apt install iproute2 gawk python3 build-essential gcc git make net-tools libncurses5-dev tftpd zlib1g-dev libssl-dev flex bison libselinux1 gnupg wget git diffstat chrpath socat xterm autoconf libtool tar unzip texinfo zlib1g-dev gcc-multilib automake screen pax gzip cpio python3-pip python3-pexpect xz-utils debianutils iputils-ping python3-git python3-jinja2 libegl1-mesa libsdl1.2-dev zlib1g:i386 libtinfo5

```

sombresleep, 2024-12-17:
@andreas-buerkler-enclustra it works fine with the prebuild file from that archive:

```
art@SS5CD4357LR3:~/enc-peta/ME-XU8-7EV-2I-D12E_PE1_SD$ petalinux-config --get-hw-description=Mercury_XU8_PE1.xsa
[INFO] Sourcing buildtools
INFO: Getting hardware description...
INFO: Renaming Mercury_XU8_PE1.xsa to system.xsa
[INFO] Generating Kconfig for project
[INFO] Menuconfig project

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.
````

It's just the XSA file that I generate from the reference design myself that isn't working. Even without making any changes (see the steps above).

sombresleep, 2024-12-17:
Update: I'm using Vivado 2024.1. Maybe I should use an older version of Vivado? 

What version of Vivado was used to generate the BSP? I see that now Vivado generates .bda file in json format, but the pre-build one from the reference design has .bda in XML.

andreas-buerkler-enclustra, 2024-12-17:
It might explain your issue in case you use the xsa file, generated from Vivado 2024.1, with an older older version of Petalinux. You can find BSP files for several versions on the release page of the corresponding reference design (https://github.com/enclustra/Mercury_XU8_PE1_Reference_Design/releases/). Recently we released reference designs for 2024.1. If you download the new bsp file and build it with Petalinux 2024.1, most probably your issue will be solved.

sombresleep, 2024-12-18:
Yep, this was the problem. Thank you for releasing the new BSPs! Now we can use the most recent AMD tools (everything worked in Vivado 2024.1 and PetaLinux 2024.1).

Now everything works as expected. I managed to enable CAN0 in Vivado, export in as an XSA file, and then open it with petalinux-config. The kernel was build without any problems, can0 link is up.

closed 2024-12-18 by sombresleep